kclyu/rpi-webrtc-streamer #86: video call with raspberry

blademckain, 2019-05-07:
it i s possibile with this package start a video call with raspberry?
from my pc i see the raspicam and ear the raspberry mic
but is possible ear in the raspberry speaker what i say in my PC microphone?

mintu19, 2019-05-08:
yes, that's what happens with this package, 2 way audio and video.

blademckain, 2019-05-08:
i have installed the package and visit http://myip:8889/native-peerconnection/ from the PC i see the raspicam stream.
1) but how do I make the pc's microphone be sent to the raspberry speaker?
2) i have installed the AppRTCMobile.apk in my phone but if i put the ip of the raspberry in the room name i get an error:
SetSDP error failed to set remote offer sdp: Session error code: ERROR_CONTENT. Session error description : Failed to set remote video description send parameters..

kclyu, 2019-05-09:
This Repo does not have much HowTo documentation for setting up or setting up audio step by step. If you do googling, you will find it enough because many people have made WebRTC related settings or other necessary things. There are not many guides and other configuration documents in this repo, but there are some who will let you know if you have questions about the problem that googling does not answer.

The AppRTCMobile question can only be answered if you tell us about the AppRTCMobile installation version.

blademckain, 2019-05-10:
ok thanks, will try to set up the audio alone. But I wanted to ask you, if the project already provides that the audio coming to the raspberry is sent to the speaker or i need to write C code for this

I am using the apk in your repo : https://github.com/kclyu/rpi-webrtc-streamer/blob/master/misc/AppRTCMobile.apk

I tried on 2 different phones with the same result.
I attach the raspberry log after pressing "call" in the phone
[log_from_app.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/3167044/log_from_app.txt)

kclyu, 2019-05-10:
No special error was found in the log file. Actually, DirectRTC detects the closed state of the TCP socket and there is nothing but the termination. 
It seems that AppRTCMobile has terminated TCP session with 'SetSDP error failed to set remote offer sdp: Session ...'  error message.

If possible, please show Offer SDP message created via [Munge-Sdp](https://webrtc.github.io/samples/src/content/peerconnection/munge-sdp/) with Android Chrome.
(Offer SDP is displayed in Offer SDP Text box when 'Get Media' -> 'Create peerconnection' -> 'Create Offer' is done.)

RWS uses ALSA. Below is a document created by Adafruit.
[USB Audio Cards with a Raspberry Pi](https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/instructions)

blademckain, 2019-05-10:
This is the offerta SDP from Android Chrome
v=0
o=- 8776362413610263960 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0 1 2
a=msid-semantic: WMS 8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG
m=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 0 8 105 13 110 113 126
c=IN IP4 0.0.0.0
a=rtcp:9 IN IP4 0.0.0.0
a=ice-ufrag:ENWU
a=ice-pwd:X0ISvctKkmb3DvT1wUrjsZ7s
a=ice-options:trickle
a=fingerprint:sha-256 8A:F3:21:B8:5E:2F:6F:98:C9:30:84:43:74:62:01:35:69:7E:E7:C5:5F:1A:26:F2:2F:CE:E7:DD:3E:67:7C:95
a=setup:actpass
a=mid:0
a=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level
a=extmap:2 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
a=extmap:3 urn:ietf:params:rtp-hdrext:sdes:mid
a=extmap:4 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id
a=extmap:5 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id
a=sendrecv
a=msid:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG 1a4f9565-8966-4bbf-a99f-d85fc9d1a4a8
a=rtcp-mux
a=rtpmap:111 opus/48000/2
a=rtcp-fb:111 transport-cc
a=fmtp:111 minptime=10;useinbandfec=1
a=rtpmap:103 ISAC/16000
a=rtpmap:9 G722/8000
a=rtpmap:0 PCMU/8000
a=rtpmap:8 PCMA/8000
a=rtpmap:105 CN/16000
a=rtpmap:13 CN/8000
a=rtpmap:110 telephone-event/48000
a=rtpmap:113 telephone-event/16000
a=rtpmap:126 telephone-event/8000
a=ssrc:1499086084 cname:ArLbNbaw9a+fPFMF
a=ssrc:1499086084 msid:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG 1a4f9565-8966-4bbf-a99f-d85fc9d1a4a8
a=ssrc:1499086084 mslabel:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG
a=ssrc:1499086084 label:1a4f9565-8966-4bbf-a99f-d85fc9d1a4a8
m=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102
c=IN IP4 0.0.0.0
a=rtcp:9 IN IP4 0.0.0.0
a=ice-ufrag:ENWU
a=ice-pwd:X0ISvctKkmb3DvT1wUrjsZ7s
a=ice-options:trickle
a=fingerprint:sha-256 8A:F3:21:B8:5E:2F:6F:98:C9:30:84:43:74:62:01:35:69:7E:E7:C5:5F:1A:26:F2:2F:CE:E7:DD:3E:67:7C:95
a=setup:actpass
a=mid:1
a=extmap:14 urn:ietf:params:rtp-hdrext:toffset
a=extmap:13 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
a=extmap:12 urn:3gpp:video-orientation
a=extmap:2 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
a=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay
a=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type
a=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing
a=extmap:8 http://tools.ietf.org/html/draft-ietf-avtext-framemarking-07
a=extmap:9 http://www.webrtc.org/experiments/rtp-hdrext/color-space
a=extmap:3 urn:ietf:params:rtp-hdrext:sdes:mid
a=extmap:4 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id
a=extmap:5 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id
a=sendrecv
a=msid:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG 41c6e22c-9cd8-4128-9924-757b48ae6fe1
a=rtcp-mux
a=rtcp-rsize
a=rtpmap:96 VP8/90000
a=rtcp-fb:96 goog-remb
a=rtcp-fb:96 transport-cc
a=rtcp-fb:96 ccm fir
a=rtcp-fb:96 nack
a=rtcp-fb:96 nack pli
a=rtpmap:97 rtx/90000
a=fmtp:97 apt=96
a=rtpmap:98 VP9/90000
a=rtcp-fb:98 goog-remb
a=rtcp-fb:98 transport-cc
a=rtcp-fb:98 ccm fir
a=rtcp-fb:98 nack
a=rtcp-fb:98 nack pli
a=fmtp:98 profile-id=0
a=rtpmap:99 rtx/90000
a=fmtp:99 apt=98
a=rtpmap:100 red/90000
a=rtpmap:101 rtx/90000
a=fmtp:101 apt=100
a=rtpmap:102 ulpfec/90000
a=ssrc-group:FID 72129130 4039669727
a=ssrc:72129130 cname:ArLbNbaw9a+fPFMF
a=ssrc:72129130 msid:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG 41c6e22c-9cd8-4128-9924-757b48ae6fe1
a=ssrc:72129130 mslabel:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG
a=ssrc:72129130 label:41c6e22c-9cd8-4128-9924-757b48ae6fe1
a=ssrc:4039669727 cname:ArLbNbaw9a+fPFMF
a=ssrc:4039669727 msid:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG 41c6e22c-9cd8-4128-9924-757b48ae6fe1
a=ssrc:4039669727 mslabel:8odGY95IkFzfsSoLstYfg2Qzma5RNbkUBHNG
a=ssrc:4039669727 label:41c6e22c-9cd8-4128-9924-757b48ae6fe1
m=application 9 DTLS/SCTP 5000
c=IN IP4 0.0.0.0
a=ice-ufrag:ENWU
a=ice-pwd:X0ISvctKkmb3DvT1wUrjsZ7s
a=ice-options:trickle
a=fingerprint:sha-256 8A:F3:21:B8:5E:2F:6F:98:C9:30:84:43:74:62:01:35:69:7E:E7:C5:5F:1A:26:F2:2F:CE:E7:DD:3E:67:7C:95
a=setup:actpass
a=mid:2
a=sctpmap:5000 webrtc-datachannel 1024

blademckain, 2019-05-10:
and this is the adb log from the AppRTCMobile where found the error
[log_apprtcmobile.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/3167963/log_apprtcmobile.txt)

blademckain, 2019-05-11:
seems an issue with h264 codec...
It is possible force rpi-webrtc-streamer to use VP8/VP9 instead oh H264 ?

kclyu, 2019-05-11:
The codec issue is correct. At present, Android OS supports H.264 hardware acceleration only in EXYNOS and QCOM GPU chip-set.  Until recently, field-trial support for MediaTek-related chip-sets has been deprecated, and now the range of supported chip-sets is reduced.

Currently, adding VP8 Codec support is considered, but all the parts related to frame processing need to be modified. Therefore, even if the change is made, it seems that it will be a long period of time. Even if V8 codec is added, quality that was supported by H.264 is not possible . It will probably be up to 640x480.

closed 2020-10-28 by kclyu