kclyu/rpi-webrtc-streamer #90: Vertical white line in video

leofds, 2019-07-13:
Hi Lyu,
Changing the background color of the "native-peerconnection" example I can see a vertical white line on the right of the video.

[image]

This does not occur using the raspistill command to take a picture.
Is this just with me ? How can I solve this ?

ronb127, 2020-10-06:
See: https://www.raspberrypi.org/forums/viewtopic.php?t=227701
It is a problem with raspberry pi's camera default ROI.
You can solve it by shrinking the ROI (add somewhere in mmal_video.c):

PARAM_FLOAT_RECT_T roi = {0.0, 0.0, 0.998, 1.0};
raspicamcontrol_set_ROI(camera->output[0], &roi);

leofds, 2020-10-09:
Thanks, I will close this issue because I can't test at the moment.

closed 2020-10-09 by leofds