kclyu/rpi-webrtc-streamer #98: Error when Building WebRTC native-code

miguelwon, 2020-05-02:
I'm getting the following error at step 3.2 of[ Building WebRTC native-code](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md#32-building-webrtc-native-code-library), and when executing `gn gen out/arm_build`:

```
mwon@mwon:/mwon_disk/disk2/raspberry/build-webrtc/src$ gn gen out/arm_build
ERROR at //third_party/BUILD.gn:30:21: Can't load input file.
    public_deps = [ "//third_party/libjpeg:libjpeg" ]
                    ^------------------------------
Unable to load:
  /mwon_disk/disk2/raspberry/build-webrtc/src/third_party/libjpeg/BUILD.gn
I also checked in the secondary tree for:
  /mwon_disk/disk2/raspberry/build-webrtc/src/build/secondary/third_party/libjpeg/BUILD.gn
```

kclyu, 2020-05-02:
This is because there was a jpeg related commit in the WebRTC native code package. The jpeg patch has not been applied to the rws master yet.

Since then, the patch has been applied to the current local master, but there is a bug in the WebRTC native code package or a problem that an additional API has not been applied, so it operates only at a specific commit position.

To use the current master brach, please use the comit position indicated at https://github.com/kclyu/rpi-webrtc-streamer/commit/8ff2e5d6a92fdfd70b19d38def379b5833362467.

kclyu, 2020-05-04:
I think, i failed to make it clear.  Rather than using the RWS commit mentioned at https://github.com/kclyu/rpi-webrtc-streamer/commit/8ff2e5d6a92fdfd70b19d38def379b5833362467, it meant moving the commit position of the WebRTC native code to that location and doing the build again.

```
cd ~/Workspace/webrtc/src
gclient sync -n -D -r 64f1f3f04edd945c8ae9f403f37ffb5cb6e7c14d
gclient sync
gn gen out/arm_build
ninja -C out/arm_build
```

SourceCodeZone, 2020-05-04:
So do I need to run below command before running above command

```
cd ~/Workspace/webrtc
fetch --nohooks webrtc
gclient sync
git config branch.autosetupmerge always
git config branch.autosetuprebase always
cd src
git checkout master
```

kclyu, 2020-05-04:
You need the WebRTC code package to go to a specific commit location, so it is correct to fetch native code at first.

kclyu, 2020-05-04:
Anyway, jpeg or pulse related include error can be solved by adding libjpeg and libpulse to rpi_rootfs below. It will build normally up to RWS, but the current master does not work with the latest WebRTC native code.

```
cd ~/Workspace/rpi_rootfs
sudo ./build_rootfs.sh cmd "/usr/bin/apt install -y libjpeg9-dev libpulse-dev"
```

SourceCodeZone, 2020-05-04:
Ok thanks for the valuable information.  I have one more doubt, I have to access the rasberryPi camera publicly without port-forwarding, I found the webrtc can use in such a situation for peer to peer communication. Like my RPi will be in local network and a webserver is installed in Godaddy with html page whch should display the live feed from Rpi. Is this possible with rpi-webrtc-streamer

kclyu, 2020-05-04:
@SourceCodeZone The content of the question is not related to this issue. In browser, WebRTC can be used as http in local network. However, in remote network (ie public network), you need to use https to use WebRTC.

SourceCodeZone, 2020-05-04:
Sorry for asking anyway my doubt is clear, I have one more issue, the command `ninja -C out/arm_build` completed without error. But when I run the command 

```
cd ~/Workspace/rpi-webrtc-streamer/src
make
```

Got  following error. 

 Linking C executable cmTC_84429
    /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_84429.dir/link.txt --verbose=1
    /opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc --sysroot=/home/user/Workspace/rpi_rootfs/rootfs    -Wl,-rpath-link=/home/user/Workspace/rpi_rootfs/rootfs/lib/arm-linux-gnueabihf:/home/user/Workspace/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -ldl  -rdynamic CMakeFiles/cmTC_84429.dir/testCCompiler.c.o  -o cmTC_84429 
    /home/user/Workspace/rpi_rootfs/tools/gcc-linaro-8.3.0-2019.03-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /home/user/Workspace/rpi_rootfs/tools/gcc-linaro-8.3.0-2019.03-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld)
    collect2: error: ld returned 1 exit status
    CMakeFiles/cmTC_84429.dir/build.make:103: recipe for target 'cmTC_84429' failed
    make[2]: *** [cmTC_84429] Error 1
    make[2]: Leaving directory '/home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeTmp'
    Makefile:138: recipe for target 'cmTC_84429/fast' failed
    make[1]: *** [cmTC_84429/fast] Error 2
    make[1]: Leaving directory '/home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeTmp'
    
    

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:180 (project)

-- Configuring incomplete, errors occurred!
See also "/home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeOutput.log".
See also "/home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeError.log".
ude -I/opt/rpi_rootfs/rootfs/opt/vc/include/interface/mmal -I/opt/rpi_rootfs/rootfs/usr/include -I/home/user/Workspace/webrtc/src/third_party/abseil-cpp/  -c websocket_server.cc -o websocket_server.o
In file included from websocket_server.h:42,
                 from websocket_server.cc:37:
websocket_server_internal.h:39:10: fatal error: libwebsockets.h: No such file or directory
 #include "libwebsockets.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:96: recipe for target 'websocket_server.o' failed
make: *** [websocket_server.o] Error 1

kclyu, 2020-05-04:
libwebsockets used by rws uses the lib/libwebsockets-3.1.zip source file in rws repo. Be sure to delete libwebsockets-dev installed in rpi_rootfs.

SourceCodeZone, 2020-05-04:
I remove it and run make again still having the first error.

kclyu, 2020-05-04:
Is your OS not Ubuntu?
Currently, it is not confirmed whether it works on Windows or Mac OS. It may be seen as an OS difference, but if you are using Ubuntu OS, check the result of the command below.

```
# cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
# make -version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
# cmake -version
cmake version 3.10.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

SourceCodeZone, 2020-05-04:
Yes Ubuntu 16.04

```
# cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.5 LTS"

# make -versionmake -versionmake -versionmake -versionmake -versionmake -versionmake -versionmake -versionmake -versionmake -version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

#  cmake -version
cmake version 3.17.0-rc1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

kclyu, 2020-05-05:
I think there is a problem building libwebsockets. Please do as below.
```
cd ~/Workspace/rpi-webrtc-streamer/lib
rm -fr libwebsockets
 ../mk/config_libwebsockets.sh /opt/rpi_rootfs
cd ../src
make
```
If the same problem persists, please check if there are any files related to libwebsockets in rpi_rootfs/rootfs.

```
cd ~/Workspace/rpi_rootfs/rootfs
sudo find . -name \*libwebsocket\*
```

SourceCodeZone, 2020-05-05:
`$../mk/config_libwebsockets.sh /opt/rpi_rootfs`
Give the output

```
start building libwebsockets library
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc - broken
CMake Error at /usr/local/share/cmake-3.17/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_9a232/fast && /usr/bin/make -f CMakeFiles/cmTC_9a232.dir/build.make CMakeFiles/cmTC_9a232.dir/build
    make[1]: Entering directory '/home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_9a232.dir/testCCompiler.c.o
    /opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc --sysroot=/home/user/Workspace/rpi_rootfs/rootfs -DBOOST_COROUTINES_NO_DEPRECATION_WARNING   -o CMakeFiles/cmTC_9a232.dir/testCCompiler.c.o   -c /home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_9a232
    /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_9a232.dir/link.txt --verbose=1
    /opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc --sysroot=/home/user/Workspace/rpi_rootfs/rootfs    -Wl,-rpath-link=/home/user/Workspace/rpi_rootfs/rootfs/lib/arm-linux-gnueabihf:/home/user/Workspace/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -ldl  -rdynamic CMakeFiles/cmTC_9a232.dir/testCCompiler.c.o  -o cmTC_9a232 
    /home/user/Workspace/rpi_rootfs/tools/gcc-linaro-8.3.0-2019.03-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /home/user/Workspace/rpi_rootfs/tools/gcc-linaro-8.3.0-2019.03-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld)
    collect2: error: ld returned 1 exit status
    CMakeFiles/cmTC_9a232.dir/build.make:103: recipe for target 'cmTC_9a232' failed
    make[1]: *** [cmTC_9a232] Error 1
    make[1]: Leaving directory '/home/user/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/CMakeFiles/CMakeTmp'
    Makefile:138: recipe for target 'cmTC_9a232/fast' failed
    make: *** [cmTC_9a232/fast] Error 2

```
And the command 
`$sudo find . -name \*libwebsocket\*`

```
./var/lib/dpkg/info/libwebsockets8:armhf.symbols
./var/lib/dpkg/info/libwebsockets8:armhf.triggers
./var/lib/dpkg/info/libwebsockets8:armhf.list
./var/lib/dpkg/info/libwebsockets8:armhf.md5sums
./var/lib/dpkg/info/libwebsockets8:armhf.shlibs
./usr/share/doc/libwebsockets8
./usr/share/lintian/overrides/libwebsockets8
./usr/lib/arm-linux-gnueabihf/libwebsockets.so.8
```

I am using latest image for rootfs `2020-02-13-raspbian-buster.img` could that be the problem.

kclyu, 2020-05-05:
The problem is GLIBC_2.27. I think you should upgrade as Ubuntu 18.04.

SourceCodeZone, 2020-05-05:
At the moment I couldn't, my server and most of the project working on Ubuntu 16.04, is there any way to GLIBC_2.27 in 16.04. Also one doubt the GLIBC_2.27 problem is in cross compiler or in my Ubuntu OS

kclyu, 2020-05-05:
you should find and install the cross compiler for raspberry pi for Ubuntu 16.04. For now, you can find it at https://github.com/abhiTronix/raspberry-pi-cross-compilers or by googling.

kclyu, 2020-05-09:
you need to add libpulse and libjpeg to rpi_rootfs as below.
```
cd ~/Workspace/rpi_rootfs
sudo ./build_rootfs.sh cmd "/usr/bin/apt install -y libjpeg9-dev libpulse-dev"
```
Please update the native code to the latest version as shown below.
```
cd ~/Workspace/webrtc/src
git pull
gclient sync
```

Also update rws to the latest master and build native code and rws as shown below.

```
cd rpi-webrtc-streamer
git pull
cp misc/webrtc_arm_build_args.gn ~/Workspace/webrtc/src/out/arm_build
cd ~/Workspace/webrtc/src
gn clean out/arm_build
gn gen out/arm_build
ninja -C out/arm_build
cd ~/Workspace/rpi-webrtc-streamer/src
make clean
make
```

closed 2020-10-28 by kclyu