kclyu/rpi-webrtc-streamer #99: Build failed

SourceCodeZone, 2020-05-02:
I am following the instruction here https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md to compile for raspberryPi(arch=armv7l
) in Ubuntu 16.04

But getting following error after running the command 

ninja -C out/arm_build

.```
./../modules/audio_coding/neteq/decision_logic.cc:355:32: warning: operation on ‘cur_size_samples’ may be undefined [-Wsequence-point]
             ? cur_size_samples = span_samples_in_packet_buffer
               ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[620/3048] CXX obj/modules/audio_device/audio_device_impl/audio_device_pulse_linux.o
FAILED: obj/modules/audio_device/audio_device_impl/audio_device_pulse_linux.o 
arm-linux-gnueabihf-g++ -MMD -MF obj/modules/audio_device/audio_device_impl/audio_device_pulse_linux.o.d -DWEBRTC_ENABLE_LINUX_ALSA -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=3fcc1d4e44127006318371002a0f421a4fde2ab4 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_ARCH_ARM -DWEBRTC_ARCH_ARM_V7 -DWEBRTC_HAS_NEON -DWEBRTC_LIBRARY_IMPL -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../.. -Igen -I../../third_party/abseil-cpp -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv7-a -mfloat-abi=hard -mtune=cortex-a7 -mfpu=neon -mthumb -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-comments -Wno-packed-not-aligned -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -gdwarf-3 -g2 -fvisibility=hidden -std=gnu++14 -Wno-narrowing -Wno-class-memaccess -fno-exceptions -fno-rtti --sysroot=../../../../../../../opt/rpi_rootfs/rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../modules/audio_device/linux/audio_device_pulse_linux.cc -o obj/modules/audio_device/audio_device_impl/audio_device_pulse_linux.o
In file included from ../../modules/audio_device/linux/audio_device_pulse_linux.h:20,
                 from ../../modules/audio_device/linux/audio_device_pulse_linux.cc:11:
../../modules/audio_device/linux/audio_mixer_manager_pulse_linux.h:14:10: fatal error: pulse/pulseaudio.h: No such file or directory
 #include <pulse/pulseaudio.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
[625/3048] CXX obj/modules/audio_device/audio_device_impl/audio_device_alsa_linux.o
ninja: build stopped: subcommand failed.

``` 

I found that inside file ~/Workspace/webrtc/src/out/arm_build/args.gn
The option 
`rtc_include_pulse_audio=false`
I think it's for turn off pulse audio. But not sure why the error happens.

kclyu, 2020-05-02:
If you are using the latest WebRTC native code package, you need to install libpulse-dev in rpi_rootfs. This is because the libpulse include file in WebRTC native code is forced to be included regardless of the args.gn setting.

Currently, the RWS master branch has problems working with the latest WebRTC native code package. 
#89 
Please use the commit position specified below for the WebRTC native code package. https://github.com/kclyu/rpi-webrtc-streamer/commit/8ff2e5d6a92fdfd70b19d38def379b5833362467

SourceCodeZone, 2020-05-04:
Thanks for the feedback, I have installed the libpulse-dev by modifying the  file `rpi_rootfs/scripts/update_upgrade_install_package.sh` with  `apt install -y  gmodule-2.0 gtk+-3.0 libasound2-dev libavahi-client-dev libpulse-dev`.

Then I download rpi-webrtc-streamer to `~/Workspace/rpi-webrtc-stremer `https://github.com/kclyu/rpi-webrtc-streamer/tree/8ff2e5d6a92fdfd70b19d38def379b5833362467 

and run the command `gn gen out/arm_build` which give me the error

```
gn gen out/arm_build
ERROR at //third_party/BUILD.gn:30:21: Can't load input file.
    public_deps = [ "//third_party/libjpeg:libjpeg" ]
                    ^------------------------------
Unable to load:
  /home/user/Workspace/webrtc/src/third_party/libjpeg/BUILD.gn
I also checked in the secondary tree for:
  /home/user/Workspace/webrtc/src/build/secondary/third_party/libjpeg/BUILD.gn
```
And when I check the folder `~/Workspace/webrtc/src/third_party/` I cannot see `libjpeg` folder but instead `libjpeg_turbo`

Also few question about https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md

The command `cp ~/Workspace/rpi-webrtc-stremer/misc/webrtc_armv6_build_args.gn arm/armv6_build/args.gn` should the directory name `rpi-webrtc-stremer` is to be `rpi-webrtc-streamer`(letter a mssing) and the output path to be `out/armv6_build/args.gn` instead `arm/armv6_build/args.gn`. Please correct me if I am wrong.

kclyu, 2020-05-04:
you are right, "out/arm_build" is corrent, not "out/arm_build"

kclyu, 2020-05-04:
Duplicate of #98

closed 2020-05-04 by kclyu

SourceCodeZone, 2020-05-04:
Thanks for replay back, I have one more doubt, I have to access the rasberryPi camera publicly without port-forwarding,  I found the webrtc can use in such a situation for peer to peer communication. Like my RPi will be in local network and a webserver is installed in Godaddy with html page whch should display the live feed from Rpi. Is this possible with rpi-webrtc-streamer