kclyu/rpi-webrtc-streamer #100: Change codec to VP8

AlexanderShniperson, 2020-05-20:
Hi
Is it possible to change Codec from current H.264 to VP8 ?

AlexanderShniperson, 2020-05-20:
i look at the source, that is impossible, maybe someone can suggest local based streaming server that can handle H.264 stream and encode them to client supported codec?

kclyu, 2020-05-20:
With raspberry pi CPU performance, vp8 is encoded at a low fps of 640x480. Decoding h.264 and encoding it as vp8 again puts a load on cpu performance.

If you need to encode with vp8, you can get it as a raw frame and encode it with vp8, but I can't recommend using it in raspberry pi. As mentioned above, video fps and delay are not good.

AlexanderShniperson, 2020-05-20:
maybe you can suggest external republishing server that can do that?

kclyu, 2020-05-20:
There is a way to use a transcoding component(MCU), but it is quite expensive to use and deploy.  It seems difficult to say that a particular product or opensource package is best.

If you need to use vp8, I think it is better to use hardware that can accelerate the encoding of vp8 codec or CPU performance.

By the way, is there any reason to use vp8?

AlexanderShniperson, 2020-05-20:
yes reason is: Android does not decode H264 frames that came from WebRTC, for test i use Chrome web browser.
readings from internet suggest to use VP8 for Android

kclyu, 2020-05-20:
Android also does h.264 decoding, it is because the smart phone model that uses a specific GPU does not support h.264. Actually, there are not many such smart phone models, but these models create the same issue.

The performance of the PI4 has improved a lot, and later models will have better CPU performance, so even if the quality is slightly deteriorated, it is time to consider the vp8 software encoding.

AlexanderShniperson, 2020-05-20:
i try to use your streamer [in my project](https://youtu.be/tUWAflE78kE) but in case of small supported percent of Android devices i rollback to MJPEG-streamer

kclyu, 2020-05-20:
Most Android models support h.264 codec. Some Android models do not support it. 
Even using Software VP8 encoding, it never achieves performance similar to hardware encoding. So, RWS didn't include VP8 codec, and even later, PI4 or the above PI version could not show hardware encoding performance.

I don't know how much MJPEG latency comes out, but for me, I would use rws for supported hardware and mjpeg for unsupported models.

Whatever you use, it's your decision. It's a good idea to use what suits your purpose.

closed 2020-10-28 by kclyu