kclyu/rpi-webrtc-streamer #103: Error 404

KrifaYounes, 2020-07-11:
When I try to access to http://IP
[image]
:8889/native-peerconnection
peer connection not working
I have followed your instructions : 
step 1 : 
docker pull kclyu/rpi-webrtc-streamer
step2 :
docker container run --device=/dev/vcsm --device=/dev/vchiq --net=host --mount type=bind,source=/var/run/dbus,target=/var/run/dbus --rm -d kclyu/rpi-webrtc-streamer # note1

How can I access raspberry streaming ? 

I have two other questions : 

Your solution can be used to control a drone from a smartphone ? 
we need to have low latency in streaming video

kclyu, 2020-07-13:
You need to use http://IP-ADDRESS:8889/native-peerconnection/. This happens because trailing / is missing in url.

closed 2020-07-13 by kclyu

kclyu, 2020-07-13:
It's another issue, and it's difficult to respond, so I'll tell you my thoughts even if it's late.

As you already mentioned, latency is the most important thing to use in drones. 
The reason why I could not answer separately is whether it is available latency or not, and it is most important that the drone you are trying to make meet the required latency.

Currently, rws is a bit faster than the video latency uploaded on youtube, but in general, the drone is not a fixed camera angle, so it requires a high bitrate even at the same resolution, keeps moving, and the wireless transmission distance continues to get farther away. I think it will also affect. 
If the evaluation is completed and please let us know, I think it will be useful for users who have the same issue.