kclyu/rpi-webrtc-streamer #104: Help streaming audio from browser/app to RWS

jonspieg, 2020-07-12:
Streaming from RWS to browser/app works great. However I can't seem to figure out how to make the other direction work as well.
I'm using the [ReSpeaker 2-mics](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/) audio hat from seeed studio and I have no problem playing audio files with it using aplay.
I've followed https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_audio.md and made sure this card&device are set as defaults in asoundrc.
I wasn't sure what I was supposed to do in /lib/modprobe.d/aliases.conf
I must say that it's not clear to me if this is something RWS even supports, the only pieces of information are vague posts here on the issues page.

Any help would be greatly appreciated
Thanks!

kclyu, 2020-07-13:
If you setup the mic capture normally in RWS, it seems that there is no problem in setting it for audio play and capture. 
In the RWS,  the part related to mic or audio depends entirely on the alsa config, and there is no enable/disable part in rws source code.

First, I assuming that you have set it up normally. To capture the audio in the browser, you must use **https**.
If you access video or audio device from browser, capture is not even possible without using https. However, in the app, I think it can be used without https.

jonspieg, 2020-07-29:
it works great from android. thanks!

closed 2020-07-29 by jonspieg