kclyu/rpi-webrtc-streamer #106: Unable to use USB Camera

h4r0ld2020, 2020-07-20:
Thanks for this wonderful program. I looked several programs on google on how I can able to live stream my USB Camera using RPI 4 B and ended up disappointed. I use motion, motioneye os, uv4l (I can't even use the UVC driver but on the demo it has a huge delays and frame drops) and other open source software. Then google pointed me to your code. I hope your code support USB camera or if you can teach me how to make it though.

kclyu, 2020-07-29:
I think you already know that the current master version does not support USB camera. USB camera support is only possible with v4l2-related video encoding.

At this time, I can't tell you when it will be supported right now, and since the raspberry pi hardware performance is improving, I am thinking of implementing it when I get some CPU performance (streaming 640x480 vp8 in real time, <500ms).

closed 2020-07-29 by kclyu

r3a1d3a1, 2021-07-10:
https://github.com/phoboslab/jsmpeg#example-setup-for-streaming-raspberry-pi-live-webcam