kclyu/rpi-webrtc-streamer #107: changing video orientation while streaming

jonspieg, 2020-07-29:
Is there a way to control video_hflip and video_vflip on the fly?
My robot changes orientation and I would like to control the streamed video orientation depending on the on-board gyroscope

Thanks

kclyu, 2020-07-29:
Please refer to http://your.pi.ip.address:8889/np2/ for changeable settings of video streaming. When streaming is activated,'show setting' menu is activated in the lower right corner, and possible video setting items will appear.
These settings can be changed even during streaming.

The API will only work with the currently connected client peer, but I think you should think about how to change the video orientation using the onboard gyroscope value of the device.

jonspieg, 2020-08-03:
the np2 example is exactly what i needed, thanks!

closed 2020-08-03 by jonspieg