kclyu/rpi-webrtc-streamer #111: Make rpi-webrtc-streamer/src  Error

mikeqoo1, 2020-09-25:
### build webrtc
```
mike@mike-linux:~/Workspace/webrtc/src (master)$ ninja -C out/arm_build
ninja: Entering directory `out/arm_build'
[1/11] LINK ./stun_prober
FAILED: stun_prober 
python "../../build/toolchain/gcc_link_wrapper.py" --output="./stun_prober" -- arm-linux-gnueabihf-g++ -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse-ld=gold -B/opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin -Wl,--threads -Wl,--thread-count=4 -Wl,--icf=all -Wl,-O2 -Wl,--gc-sections -Wl,--gdb-index --sysroot=../../../../../../../opt/rpi_rootfs/rootfs -L../../../../../../../opt/rpi_rootfs/rootfs/opt/vc/lib -L../../../../../../../opt/rpi_rootfs/rootfs/usr/local/lib/arm-linux-gnueabihf -L../../../../../../../opt/rpi_rootfs/rootfs/lib/arm-linux-gnueabihf -L../../../../../../../opt/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -L../../../../../../../opt/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf/libfakeroot -L../../../../../../../opt/rpi_rootfs/rootfs/usr/local/lib -pie -Wl,--disable-new-dtags -o "./stun_prober" -Wl,--start-group @"./stun_prober.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -ljpeg -lX11
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/libstdc++.so: error: undefined reference to 'log', version 'GLIBC_2.29'
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/libstdc++.so: error: undefined reference to 'pow', version 'GLIBC_2.29'
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/libstdc++.so: error: undefined reference to 'exp', version 'GLIBC_2.29'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
### make src
```
mike@mike-linux:~/Workspace/rpi-webrtc-streamer/src (master)$ make
arm-linux-gnueabihf-g++  -I. -DINSTALL_DIR=\"/opt/rws\" -Wno-deprecated-declarations -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv7-a -mfloat-abi=hard -mtune=cortex-a7 -mfpu=neon -mthumb -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-comments -Wno-packed-not-aligned -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g2 -fvisibility=hidden -isystem/opt/rpi_rootfs/rootfs/usr/include/gtk-3.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/at-spi2-atk/2.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/at-spi-2.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/dbus-1.0 -isystem/opt/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -isystem/opt/rpi_rootfs/rootfs/usr/include/gtk-3.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/gio-unix-2.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/cairo -isystem/opt/rpi_rootfs/rootfs/usr/include/pango-1.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/harfbuzz -isystem/opt/rpi_rootfs/rootfs/usr/include/pango-1.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/fribidi -isystem/opt/rpi_rootfs/rootfs/usr/include/atk-1.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/cairo -isystem/opt/rpi_rootfs/rootfs/usr/include/pixman-1 -isystem/opt/rpi_rootfs/rootfs/usr/include/freetype2 -isystem/opt/rpi_rootfs/rootfs/usr/include/libpng16 -isystem/opt/rpi_rootfs/rootfs/usr/include/gdk-pixbuf-2.0 -isystem/opt/rpi_rootfs/rootfs/usr/include/libmount -isystem/opt/rpi_rootfs/rootfs/usr/include/blkid -isystem/opt/rpi_rootfs/rootfs/usr/include/uuid -isystem/opt/rpi_rootfs/rootfs/usr/include/glib-2.0 -isystem/opt/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf/glib-2.0/include -DINSTALL_DIR=\"/opt/rws\" -Wall -fvisibility=hidden -pthread -std=gnu++14 -Wno-narrowing -Wno-class-memaccess -fno-exceptions -fno-rtti --sysroot=../../../../../../../opt/rpi_rootfs/rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -Wno-deprecated-declarations -Wno-deprecated-declarations -Wno-undef -D__RWS_VERSION__=\"v0.74-8-gd75d2b28dfe6beb2\" -D__WEBRTC_VERSION__="\"7a7683567cd830d3,Cr-Commit-Position:refs/heads/master@{#32189}\"" -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -I/home/mike/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build -I/home/mike/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build/include -I/home/mike/Workspace/rpi-webrtc-streamer/lib/libwebsockets -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DUSE_X11=1 -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=c2e54f675b83a61301dcdb22e8e7a2b85c01d58c -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DRTC_ENABLE_VP9 -DHAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_ARCH_ARM -DWEBRTC_ARCH_ARM_V7 -DWEBRTC_HAS_NEON -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -DABSL_FLAGS_STRIP_NAMES=0 -DHAVE_WEBRTC_VIDEO -I/home/mike/Workspace/webrtc/src -I/home/mike/Workspace/webrtc/src -I/home/mike/Workspace/webrtc/src/out/gen -I/home/mike/Workspace/webrtc/src/third_party/abseil-cpp -I/home/mike/Workspace/webrtc/src/third_party/libyuv/include -I/home/mike/Workspace/webrtc/src/third_party/jsoncpp/source/include -I/home/mike/Workspace/webrtc/src/third_party/jsoncpp/generated  -I/opt/rpi_rootfs/rootfs/opt/vc/include/interface/vcos/pthreads -I/opt/rpi_rootfs/rootfs/opt/vc/include/interface/vmcs_host/linux -I/opt/rpi_rootfs/rootfs/opt/vc/include/interface/vmcs_host -I/opt/rpi_rootfs/rootfs/opt/vc/include/interface/vchiq_arm -I/opt/rpi_rootfs/rootfs/opt/vc/include -I/opt/rpi_rootfs/rootfs/opt/vc/include/interface/mmal -I/opt/rpi_rootfs/rootfs/usr/include -I/opt/rpi_rootfs/rootfs/usr/include/arm-linux-gnueabihf -I/home/mike/Workspace/webrtc/src/third_party/abseil-cpp/  -c direct_socket.cc -o direct_socket.o
direct_socket.cc: In constructor ‘DirectSocketServer::DirectSocketServer()’:
direct_socket.cc:56:49: error: no matching function for call to ‘rtc::MessageHandler::MessageHandler()’
   56 |     : listener_(nullptr), direct_socket_(nullptr) {
      |                                                 ^
In file included from /home/mike/Workspace/webrtc/src/rtc_base/thread.h:34,
                 from /home/mike/Workspace/webrtc/src/rtc_base/net_helpers.h:28,
                 from direct_socket.h:37,
                 from direct_socket.cc:37:
/home/mike/Workspace/webrtc/src/rtc_base/message_handler.h:41:12: note: candidate: ‘rtc::MessageHandler::MessageHandler(bool)’
   41 |   explicit MessageHandler(bool auto_cleanup);
      |            ^~~~~~~~~~~~~~
/home/mike/Workspace/webrtc/src/rtc_base/message_handler.h:41:12: note:   candidate expects 1 argument, 0 provided
```

### My Version
```
mike@mike-linux:~/Workspace/webrtc/src (master)$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal
mike@mike-linux:~/Workspace/webrtc/src (master)$ ldd --version
ldd (Ubuntu GLIBC 2.31-0ubuntu9.1) 2.31

```

kclyu, 2020-09-25:
***WebRTC native code compiling***
With Custom Compiled GCC, it is currently being compiled without problems. Maybe there is a problem compiling with Ubuntu arm-linux-gnueabihf gcc-9.
```
                                   ^~~~~~~~~~~~~~~~~~~~~
[2868/2882] CXX obj/video/video/rtp_video_stream_receiver2.o
In file included from ../../modules/rtp_rtcp/source/rtp_sender_egress.h:21,
                 from ../../modules/rtp_rtcp/source/rtp_rtcp_impl2.h:36,
                 from ../../video/rtp_video_stream_receiver2.h:33,
                 from ../../video/rtp_video_stream_receiver2.cc:11:
../../api/rtc_event_log/rtc_event_log.h:49:16: warning: ‘virtual void webrtc::RtcEventLog::StopLogging(std::function<void()>)’ was hidden [-Woverloaded-virtual]
   virtual void StopLogging(std::function<void()> callback) {
                ^~~~~~~~~~~
../../api/rtc_event_log/rtc_event_log.h:63:8: warning:   by ‘virtual void webrtc::RtcEventLogNull::StopLogging()’ [-Woverloaded-virtual]
   void StopLogging() override {}
        ^~~~~~~~~~~
[2882/2882] STAMP obj/default.stamp
```

***RWS compiling***
There was a problem with RWS compilation in the latest WebRTC native code due to the recent change in MessageHandler, common_types, alsa_linux X11 dependency. Pull again, and build WebRTC native code library again by copying the args.gn file to out/arm_build.

mikeqoo1, 2020-09-25:
hi @kclyu 

i build webrtc is work but i make WebRTC native code have error 

thank you =)

[image]
[image]
[image]

kclyu, 2020-09-26:
As there is a flag called -march x86_64, it seems that WebRTC native code is compiled with x86_64 arch option, it means you did build WebRTC native code without args.gn  file. Make sure that the misc/webrtc_arm_build_args.gn file is properly copied to the arm_build/args.gn file and rebuild.

mikeqoo1, 2020-09-26:
I can’t thank you enough！！！

[image]

thx you

closed 2020-10-28 by kclyu