kclyu/rpi-webrtc-streamer #113: Select camera crop

jonspieg, 2020-10-06:
Can I somehow define the portion of the sensor I wish to stream?
Ideally I would like to define a rectangle of resolution 1280x720 starting from the center-top of the sensor.

Is this type of pixel manipulation supported in RWS and if so, how would I go about tackling this.

Thanks!

kclyu, 2020-10-06:
Currently, roi setting is not supported in config or websocket interface. In websocket interface, zoom in/out/move is supported. For an example command, refer to the function that executes sendZoomCommand in web-root/np2/js/streamer_controller.js.

It may be necessary to focus a specific area in a fixed camera configuration, but since I thought that the roi setting was not necessary in the config,  Please let me know if you have any different opinions.

Mzeron, 2020-10-08:
Hi, 
This is a very interesting topic that I will also find very useful,
Today, (from my playing around) there is no way to get the full scope of the camera sensor to go through the RWS, so a good solustion can de selecting region of interest and stream only a specific part.
You can configure to stream the upper or down part of the camera view.
Do you think it is a possible to add ROI to the config files?

Thank you for the great RWS library!

kclyu, 2020-10-16:
Camera roi setting have been added to media_config.
For more details, please refer to the commit log above.

closed 2020-10-28 by kclyu