kclyu/rpi-webrtc-streamer #114: Increased binary size (rtc_use_h264=true)

ronb127, 2020-10-17:
Hello,
In commit 4a64301c: "Source update for WebRTC Cr-Commit Position 31187" rtc_use_h264 was changed from 'false' to 'true'.
This causes an increase of 2-3 MB in binary size (ffmpeg is now compiled in webrtc).
If I build webrtc with "rtc_use_h264=false" then the binary size is reduced again, but H.264 is no longer offered in the SDP and the video stream doesn't work.
Can we somehow go back to "rtc_use_h264=false" (for smaller binary size) without disabling H.264, as it was before?
Thanks.

kclyu, 2020-10-17:
For now, it seems difficult unless you modify the webrtc native code itself. By the way, it doesn't seem that the 2~3M increase is so sensitive, do you use it for other purposes?

ronb127, 2020-10-17:
Just trying to keep my binaries small.
I wonder how it worked before, and what makes it fail now. ffmpeg is clearly unnecessary with external H.264 streams. Maybe I'll post a bug/request on webrtc forums.

ronb127, 2020-10-28:
I was able to exclude ffmpeg from the build by applying a small patch to webrtc:
modules/video_coding/BUILD.gn
modules/video_coding/codecs/h264/h264.cc
See patch file: [h264.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/5450321/h264.txt)
You should build with use_rtc_h264=true, only now ffmpeg is excluded and binary size is reduced.

kclyu, 2020-10-28:
I will refer to it later when creating a new rws package. Thank you.

closed 2020-10-28 by kclyu