kclyu/rpi-webrtc-streamer #115: Is it possible to add snapshot support?

Aaltuj, 2020-10-29:
Hi @kclyu,

Great work, I am scouting for a new webcam solution for octoprint. One of the features is making a time-lapse.  This requires a http cal that returns a still of the stream. Currently that is done with MJPEG-STREAMER and i was wondering if there is a possiblity already or in the near future to do that.

Regards,
Alex

ps: saw #96 but still had some hope ;)

kclyu, 2020-10-30:
Currently, there is no function to provide still images through http. I am going to develop a function similar to the function mentioned above in the near future(I think it will be done within this year, but at the latest at the beginning of next year), but in fact, the priority is lower than that of others, so the development is not started yet.

From what you are saying, you seem to talk about a function that actually provides time-lapse in addition to periodically capturing still images in Rpi. In fact, it seems difficult to provide a time-lapse function in RWS. It seems that you have to receive a still image and use it to make the time-lapse function yourself.

Aaltuj, 2020-10-30:
oke great to hear it's in the pipeline. The timelapse part is done by octoprint, so it only needs the frame. so no biggy there. Thanks for the response and i'll wait patiently  ;)

kclyu, 2020-10-31:
For your info, It is also possible to use the still image capture function without the RWS capture function.  RWS uses the camera device only during video streaming.  In other words, you can capture still images with raspistill command when streaming is not in use with cron or other background jobs. If you do not want to run a separate web server in Rpi, you can use the http function of RWS. The directory listing is blocked, but if the requested file is inside the web-root you can fetch the file with http.

Aaltuj, 2020-11-01:
Thanks for the suggestion, Wil dive into that.

closed 2020-11-03 by Aaltuj