kclyu/rpi-webrtc-streamer #116: Is there some way to have multiple connections?

juniorsaldanha, 2020-11-23:
Hi, i'd like to know if there is some way to have multiples connections on the same source video (raspicam TC358743), with only one connection work great, but i couldnt find an way to have multiples connections.

Here is the message that i received on web browser console:
[image]

thnks , :+1:

kclyu, 2020-11-24:
RWS itself does not support multiple rtp connections.

If you want to make multiple RTP connections, you should use it in  interworking with gateway functions such as videoroom supported by WebRTC gateway. 

Currently, I are developing a function to interwork with janus gatway's videoroom, and when it is completed, multiple RTP connections are possible.  For developments related to janus gateway development, please refer to  [Connect to Janus Gateway](#43)

closed 2020-11-24 by kclyu

gunnarsundberg, 2022-06-26:
@kclyu are there any updates on integrating with Janus?