kclyu/rpi-webrtc-streamer #117: What would be the likely issue for unable to stream on the USB Ethernet (OTG)

jasonmnemonic, 2021-03-02:
Hi.

Thank you for the great work! ;-) It is interesting to see how WebRTC work. I have downloaded the 0.74.1 deb package and installed on my Pi Zero. Two questions.

1) My Pi Zero is configured as both WiFi and USB Ethernet. For some reason, I can only stream on the WiFi e.g. open Chrome and go to `192.168.150.33:8887/np2` and note the port is 8887 because I have changed it in the config file.

On WiFi, it works but when I disabled the WiFi so that I can be sure the stream will go on the local USB ethernet, nothing works. E.g. my Pi has USB IP address of 192.168.200.1 and I can ping and SSH into it and all is good. However, I see both cases:

1.1) On Chrome, if I use `http://192.168.200.1:8887/np2` then the browser only gives a white page with a line of text showing `PeerConnection Page - RWS` and nothing else.

1.2) If I use `http://my-raspberry-pi:8887/np2` which hostname has the same IP address I can see the normal `PeerConnection Page - RWS` with the usual preview and two buttons, Connect and Disconnect. When I clicked on Connect, I get the quick few seconds pop up; `Error during connect wss://avs-pov1-abra:8887/rws/ws` but no picture.

In my head, this should be no different to WiFi since I have now disabled WiFi and it should work. Do you have any idea? I cannot see any logs in `/opt/rws/log` because they are empty and I do not know where to start looking.

2) Also is there any way to configure the stream so that I can annotate the text to show bitrate, fps and resolution? Is there a way to confirm the resolution, fps and bitrate so I can double check that I think the output that I see is correct.

Thank you! :-)

kclyu, 2021-03-03:
1.1) It looks like you need to add / after np2.
```
http://192.168.200.1:8887/np2/
```
1.2)  The attempt to connect with wss seems to be using dns. This seems to require javascript modification. Until modification, you can use the ip address or modify it yourself. 

2. The fps/bitrate/resolution can be viewed in more detail in the ***Stats graphs for RTCInboundRTPVideoStream*** at chrome://webrtc-internals/ during streaming.

jasonmnemonic, 2021-03-03:
Thank you for the reply!

OK to questions (1.1) and (2) and they work. Could you please explain (1.2); I do not understand. Is that javascript on the Pi server side?

(3) Also another question if I may, please. OK this question may or may not make sense; it is because I am trying to understand how webrtc, webrtc-streamer, and STUN & TURN work. I think I know why from my observations but unsure how to get about this or is there a way to make this setup work? I think this is related to questions (1.1 & 1.2).

I have a Pi and it has WiFi and USB Ethernet (OTG) to my Windows PC. So this means I can SSH into Pi on the USB and also elsewhere through the WiFi. The Pi's WiFi is connected to a router so it has DNS and my Pi has a proper IP that is 192.168.x.x and the desktop is also in the same sub-net.

So my intention is to have streaming to happen on the USB ethernet link than the WiFi so I have better stream speed and also I want to be able to stream without connecting to the internet. This is what I noticed:

I cannot remember if this is true because I was experimenting and also new to this. If I have WiFi disabled and I clicked on Connect, I get the quick few seconds pop up; _Error during connect wss://my-raspberry-pi:8887/rws/ws_ but no picture when WiFi is disabled. If WiFi is enabled, I can start the stream and disable WiFi in the middle of the stream and this causes the stream to stall for a few seconds before continuing on the USB ethernet link. This sorts of confirms of my use-case that data is going on the USB link.

For some reason, logging starts to work and when I try to understand the logfile (it is difficult for me but I try), I think as Connect is started, **webrtc-streamer** goes to Google's STUN & TURN servers which sorts of allow communications to go through the firewalls between both Pi and Desktop and then stream happens in which disabling WiFi then is OK.

The question is (maybe it is already happening and my observation is wrong); if I want to be able to stream without connecting to the internet, how do I get around the STUN & TURN servers since my Pi ZeroW and Desktop are both on the table and no internet? Is it possible to install local STUN & TURN servers?

(4) I also noticed sometimes after a while e.g. say after 10 minutes at 1080p at 25 FPS on the Pi ZeroW, the streamer stops streaming. When this happens, sometimes I have low CPU % and sometimes I could have near 90% CPU in the top app/utility. Not sure if it is a crash or something unexpected. Any idea?

(5) Sometimes when I have FPS 25, I can see the Chrome's framesPerSecond graph spiking within a minute between 10 to 30 FPS and sometimes between 15 and 30 and this is with both dynamic FPS and dynamic resolution disabled. Is there something to make this stable where it is nearer to 25 with 23 - 25?

Thank you again! :-)

kclyu, 2021-03-04:
1.2) To use webrtc in brower you need javascript. /opt/rws/web-root/np2/js is the javascirpt used by np2.

4,5) The webrtc bandwidth may vary depending on the situation (network transmission speed, or peer's performance). In rws, the result is a resolution change or fps decrease. It is possible that the video quality deteriorates in certain situations, and if it becomes a problem that can be reproduced, it seems that action should be taken.

It is difficult to answer all of your question that are rarely related to the rws issue. Please understand this,
Many contents related to webrtc and materials that you are curious about at the time will be able to find enough with googling.

jasonmnemonic, 2021-03-04:
Thank you for your replies. It is much appreciated.

I go and experiment and maybe Google enough and if I am stuck on the use-case, I will hopefully be able to ask direct questions and get some guidance if it is OK?

Thank you and have a good day! :-)

ny1236, 2023-10-05:
http://192.168.200.1:8887/np2/