kclyu/rpi-webrtc-streamer #121: deb package very outdated

ArosPrince, 2021-04-24:
Hi,
first of all, this is a great project. Many thanks for doing this.

Now the deb package is very outdated. More that a year old and obviously does not work with the latest changes of the js files. Can you please update it?

Also it would be nice if the project could be native-compiled on Pi. Not everybody has a setup for cross-compiling.

Thanks!

kclyu, 2021-04-25:
It was already over a year and I was planning to update it. However, there is an issue that has not yet been fixed, so I have not yet been able to update it, and I haven't been able to fix it because I haven't used a lot of time to fix the issue. If the problem is fixed, I will update it immediately.

The problem of compiling directly in raspberry pi is that WebRTC native code cannot be compiled directly in Raspberry Pi (raspios). I haven't tried it on Ubuntu for Raspberry Pi yet, but if it is possible, it seems to be possible for Raspberry Pi Ubuntu 64bit. For raspios, cross compiling is currently the only way.

ArosPrince, 2021-05-04:
OK, thanks for letting me know. Will try installing Ubuntu then. Thanks!