kclyu/rpi-webrtc-streamer #122: "/dev/vcsm": no such file or directory.

RobasaurusRex, 2021-05-01:
I'm getting this error while installing on a Raspberry Pi 4 with the newest Raspberry Pi OS:

pi@Pi4-2:~ $ sudo docker container run --device=/dev/vcsm --device=/dev/vchiq --net=host --mount type=bind,source=/var/run/dbus,target=/var/run/dbus --rm -d kclyu/rpi-webrtc-streamer # note1
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
d840ef9d534f4000a560d2acea6eb8e89faa4915f5d6a9c0bad447e3318203c8
docker: Error response from daemon: error gathering device information while adding custom device "/dev/vcsm": no such file or directory.

ThumbGen, 2021-05-07:
To fix it, you just need to replace **--device=/dev/vcsm** by **--device=/dev/vcsm-cma**

Credit: @NuAlex

RobasaurusRex, 2021-05-12:
I really, really appreciate your help.  I'll probably write a how-to for others. I'll send you a link if I do. 

Rob