kclyu/rpi-webrtc-streamer #123: Cross-compile issue

ArosPrince, 2021-05-18:
Hi, I am sorry if this is a dumb question but even after several hours I wasn't able to figure out what's wrong.

In the guide to crosscompile on Ubuntu, in step 2.2, one is supposed to run some git config commands after calling `gclient sync`

The problem is that it says it is not a git directory:

```
~/Workspace/webrtc$ git config branch.autosetupmerge always
fatal: not in a git directory
```

...which is right as far as I can tell.

Could it be there is some step left out in the cross-compile guide?

Thanks

adgbu, 2021-05-20:
I have not built rpi-webrtc-streamer yet, but I did manage to cross compile WebRTC on Ubuntu x64 for Raspberry OS ARMv7 target.

Comparing my own notes to README_building.md step 2.2.

After

    $ fetch --nohooks --no-history webrtc

If you are on Linux and fetching the code for the first time, then you need to run:

    $ cd src
    $ ./build/install-build-deps.sh

After that I continued like this.

    $ git checkout master
    $ git pull origin master
    $ gclient sync

Then prepare for build according to GN Quickstart guide
https://gn.googlesource.com/gn/+/master/docs/quick_start.md

I hope that helps. I'm also new to these tools.

ArosPrince, 2021-05-22:
Thanks. Running install-build-deps.sh prior to those git commands indeed helped.

However, later on when calling 
`./build_rootfs.sh create yyyy-mm-dd-raspios-buster.img`

(in my case it is 2021-03-04-...)

I get a lot of "permission denied":

[image]

Any more ideas what I may be doing wrong?

Thanks!

adgbu, 2021-05-24:
Sorry, I haven't done it that way. I got the raspberry pi tools and compiler from
$ git clone https://github.com/raspberrypi/tools.git

Just an idea. Did you run a command with sudo when you were not supposed to? Then user root will own the files and you (normal user) will not have permissions to access those files.

ArosPrince, 2021-05-24:
Yes, you're right again...

I have created PR with the two fixes plus another typos I found:
https://github.com/kclyu/rpi-webrtc-streamer/pull/125

I am still not finished though, so there may be other issues.

Thanks!

aphofstede, 2022-02-17:
Did you get this running on Ubuntu x64? We plan to crosscompile for arm64 but I'm a bit concerned about MMAL support: https://github.com/raspberrypi/userland/issues/688

ArosPrince, 2022-02-18:
If you're asking me... Then yes, I used the exact way described. I.e. I compiled on x64 Ubuntu (running in Hyper-V) using these steps: https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md