kclyu/rpi-webrtc-streamer #124: Building WebRTC native-code library

allen123456allen, 2021-05-19:
The following error occurred when I ran the "gn gen out/arm_build" command:
RROR at //build/config/compiler/BUILD.gn:644:5: Assertion failed.
Assert (USE_LLD, "LTO is only supported with LLD ")
^ -- -- -- -- --
LTO is only supported with lld
See //build/config/BUILDCONFIG.gn:318:3: which caused the file to be included.
"/ / build/config/compiler: afdo",
^ -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -
Can you take a look at it for me?Thank you very much!

ArosPrince, 2021-05-24:
Exactly the same issue here.

Just for reference. This is the code that is causing it, but I cannot make out anything from the description. Also googling it yields no results at all...

[image]

BTW sorry for the screenshots. I cannot seem to be able to copy text from the VM. Something wrong with the integration...

allen123456allen, 2021-05-27:
Change is_official_build=false is_debug=true in the args.gn file and the recompile will be successful.

kclyu, 2021-05-28:
There are some issues with WebRTC native code package building. It seems to be an issue mostly related to code optimization.And  It seems that it may take more time for it to stabilize. This issue will be deleted later when WebRTC native code is built with this option and the problem goes away. 

In addition to this issue, it has not been confirmed whether there is a problem with neon-related optimization or gcc issue in the recent WebRTC native code and libyuv, but I think it will be fixed after a little time.

To compile the latest WebRTC native code, you need to modify the following part in the args.gn file as well.
```
arm_use_neon=false
arm_optionally_use_neon=false
rtc_build_with_neon=false
```

ArosPrince, 2021-05-29:
After updating the five items, I am able to run `gn gen`. But the subsequent `ninja -C out/arm_build` command still fails with some neon related errors:
```

[2263/2924] LINK ./rgba_to_i420_converter
FAILED: rgba_to_i420_converter 
python3 "../../build/toolchain/gcc_link_wrapper.py" --output="./rgba_to_i420_converter" -- arm-linux-gnueabihf-g++ -Wl,--fatal-warnings -Wl,--build-id -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse-ld=gold -B/opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin -Wl,--threads -Wl,--thread-count=4 -Wl,--gdb-index -rdynamic --sysroot=../../../../../../../opt/rpi_rootfs/rootfs -L../../../../../../../opt/rpi_rootfs/rootfs/opt/vc/lib -L../../../../../../../opt/rpi_rootfs/rootfs/usr/local/lib/arm-linux-gnueabihf -L../../../../../../../opt/rpi_rootfs/rootfs/lib/arm-linux-gnueabihf -L../../../../../../../opt/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -L../../../../../../../opt/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf/libfakeroot -L../../../../../../../opt/rpi_rootfs/rootfs/usr/local/lib -pie -Wl,--disable-new-dtags -o "./rgba_to_i420_converter" -Wl,--start-group @"./rgba_to_i420_converter.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt
../../third_party/libyuv/source/convert.cc:1113: error: undefined reference to 'YUY2ToYRow_NEON'
../../third_party/libyuv/source/convert.cc:1113: error: undefined reference to 'YUY2ToUVRow_NEON'
../../third_party/libyuv/source/convert.cc:1204: error: undefined reference to 'UYVYToYRow_NEON'
../../third_party/libyuv/source/convert.cc:1204: error: undefined reference to 'UYVYToUVRow_NEON'
../../third_party/libyuv/source/convert.cc:1273: error: undefined reference to 'AYUVToYRow_NEON'
../../third_party/libyuv/source/convert.cc:1273: error: undefined reference to 'AYUVToUVRow_NEON'
../../third_party/libyuv/source/convert.cc:1342: error: undefined reference to 'AYUVToYRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:82: error: undefined reference to 'CopyRow_NEON'
../../third_party/libyuv/source/convert.cc:1342: error: undefined reference to 'AYUVToVURow_NEON'
../../third_party/libyuv/source/convert.cc:1442: error: undefined reference to 'ARGBToYRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:477: error: undefined reference to 'SplitUVRow_NEON'
../../third_party/libyuv/source/convert.cc:1442: error: undefined reference to 'ARGBToUVRow_NEON'
../../third_party/libyuv/source/convert.cc:1534: error: undefined reference to 'BGRAToYRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:552: error: undefined reference to 'MergeUVRow_NEON'
../../third_party/libyuv/source/convert.cc:1534: error: undefined reference to 'BGRAToUVRow_NEON'
../../third_party/libyuv/source/convert.cc:1636: error: undefined reference to 'ABGRToYRow_NEON'
../../third_party/libyuv/source/convert.cc:1636: error: undefined reference to 'ABGRToUVRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:609: error: undefined reference to 'SplitUVRow_16_NEON'
../../third_party/libyuv/source/convert.cc:1728: error: undefined reference to 'RGBAToYRow_NEON'
../../third_party/libyuv/source/convert.cc:1728: error: undefined reference to 'RGBAToUVRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:664: error: undefined reference to 'MergeUVRow_16_NEON'
../../third_party/libyuv/source/convert.cc:1900: error: undefined reference to 'RGB24ToYRow_NEON'
../../third_party/libyuv/source/convert.cc:1900: error: undefined reference to 'RGB24ToUVRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:714: error: undefined reference to 'MultiplyRow_16_NEON'
../../third_party/libyuv/source/convert.cc:2072: error: undefined reference to 'RGB24ToARGBRow_NEON'
../../third_party/libyuv/source/convert.cc:2072: error: undefined reference to 'ARGBToYJRow_NEON'
../../third_party/libyuv/source/convert.cc:2072: error: undefined reference to 'ARGBToUVJRow_NEON'
../../third_party/libyuv/source/convert.cc:2243: error: undefined reference to 'RAWToYRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:764: error: undefined reference to 'DivideRow_16_NEON'
../../third_party/libyuv/source/convert.cc:2243: error: undefined reference to 'RAWToUVRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:820: error: undefined reference to 'SwapUVRow_NEON'
../../third_party/libyuv/source/convert.cc:2415: error: undefined reference to 'RAWToARGBRow_NEON'
../../third_party/libyuv/source/convert.cc:2415: error: undefined reference to 'ARGBToYJRow_NEON'
../../third_party/libyuv/source/convert.cc:2415: error: undefined reference to 'ARGBToUVJRow_NEON'
../../third_party/libyuv/source/convert.cc:2573: error: undefined reference to 'RGB565ToYRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:922: error: undefined reference to 'SplitRGBRow_NEON'
../../third_party/libyuv/source/convert.cc:2573: error: undefined reference to 'RGB565ToUVRow_NEON'
../../third_party/libyuv/source/convert.cc:2735: error: undefined reference to 'ARGB1555ToYRow_NEON'
../../third_party/libyuv/source/convert.cc:2735: error: undefined reference to 'ARGB1555ToUVRow_NEON'
../../third_party/libyuv/source/convert.cc:2912: error: undefined reference to 'ARGB4444ToYRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:1057: error: undefined reference to 'SplitARGBRow_NEON'
../../third_party/libyuv/source/convert.cc:2912: error: undefined reference to 'ARGB4444ToUVRow_NEON'
[... cut ...]
../../third_party/libyuv/source/planar_functions.cc:4517: error: undefined reference to 'ByteToFloatRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:4682: error: undefined reference to 'ARGBExtractAlphaRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:4876: error: undefined reference to 'SplitUVRow_NEON'
../../third_party/libyuv/source/planar_functions.cc:5058: error: undefined reference to 'HalfMergeUVRow_NEON'
collect2: error: ld returned 1 exit status
[2268/2924] CXX obj/third_party/openh264/encoder/encoder_ext.o
ninja: build stopped: subcommand failed.
```
Any idea?

kclyu, 2021-05-31:
I already talked about what needs to be fixed in the args.gn file in the posting right above.

FYI,  when RWS uses the latest WebRTC native code, compilation errors (API changes related to Platform Thread in WebRTC, and header files related to H264 profile are changed) occur. The fix will be included in the next commit.

For changes to the WebRTC native code commit location that is compiled with RWS, please refer to [3.4 moving to specific commit position](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md#34-moving-to-specific-commit-position-of-webrtc-native-code-package)

ArosPrince, 2021-05-31:
Thanks for the info, but I included those variables you listed in args.gn. Maybe I don't fully understand what you're saying.

Anyway, yes, seems I have misunderstood that 3.4 section and I am sorry for that. Will try again.

ArosPrince, 2021-06-07:
OK, I have been able to successfully build everything based on the commit contained in the rws repo commit message. Great. One more question though... 

Upon running the streamer and connecting to the URL I get the following error:

`(app_ws_client.cc:731): Still Capture Status : INVALID_ARGUMENT: Failed to create file, error: No such file or directory`

I have already enabled the camera in raspi-config before which I was getting another error, but that went away with enabling the camera.

But I can't seem to figure out how get rid of this one.

ArosPrince, 2021-06-09:
OK, seems it's trying to save the still capture to `/opt/rws/still_captured`.

My guess is that this folder can be configured in one of the config files, right? Only I cannot find where...

ArosPrince, 2021-07-09:
Just for the record... Creating the directory helps, but you still need to run `./webrtc-streamer --verbose` with `sudo`, otherwise it fails with permission denied (obviously as it's trying to write to `/opt`)

It seems to me that is has to be configurable somewhere, but I just cannot figure out where.

icemanaf, 2021-12-18:
I'm getting a build error step 3.2 when building the native WebRTC library for arm6.
I'm using the commit 130e0310108017d32e22377306b0d02ac21d3cc6 for the library, which I got from the RWS commit log as per the build instructions. Any ideas? 

```
[2527/2935] CC obj/third_party/ffmpeg/ffmpeg_internal/vp8.o
FAILED: obj/third_party/ffmpeg/ffmpeg_internal/vp8.o
arm-linux-gnueabihf-gcc -MMD -MF obj/third_party/ffmpeg/ffmpeg_internal/vp8.o.d -DHAVE_AV_CONFIG_H -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -DFFMPEG_CONFIGURATION=NULL -D_ISOC99_SOURCE -D_LARGEFILE_SOURCE -DUSE_UDEV -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DUSE_X11=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -DCR_SYSROOT_HASH=11d6f690ca49e8ba01a1d8c5346cedad2cf308fd -D_DEBUG -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DOPUS_FIXED_POINT -I../../third_party/ffmpeg/chromium/config/Chrome/linux/arm -I../../third_party/ffmpeg -I../.. -Igen -I../../third_party/opus/src/include -fPIC -Wno-deprecated-declarations -fomit-frame-pointer -w -std=c99 -pthread -fno-math-errno -fno-signed-zeros -fno-tree-vectorize -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -march=armv6 -mfloat-abi=hard -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -mfpu=vfp -marm -g2 -gsplit-dwarf -fvisibility=hidden -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-comments -Wno-packed-not-aligned -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fdata-sections -ffunction-sections -std=gnu11 --sysroot=../../../../../../opt/rpi_rootfs/rootfs -c ../../third_party/ffmpeg/libavcodec/vp8.c -o obj/third_party/ffmpeg/ffmpeg_internal/vp8.o
{standard input}: Assembler messages:
{standard input}:2597: Error: bad instruction `ldrhcs fp,[lr],#2'
{standard input}:2796: Error: bad instruction `ldrhcs r0,[ip],#2'
{standard input}:2900: Error: bad instruction `ldrhcs r9,[ip],#2'
{standard input}:2989: Error: bad instruction `ldrhcs r4,[ip],#2'
{standard input}:3174: Error: bad instruction `ldrhcs r9,[lr],#2'
{standard input}:3254: Error: bad instruction `ldrhcs r0,[lr],#2'
{standard input}:3314: Error: bad instruction `ldrhcs r9,[lr],#2'
{standard input}:3405: Error: bad instruction `ldrhcs r9,[lr],#2'
```

jasonmnemonic, 2022-04-11:
Hi.

Instead of duplicating my post which I have done on another closed post (not sure if closed means the notification stops working); I have the same problem with different error messages. Yes, I also followed @kclyu and @allen123456allen posts on args.gn on neon and also using debug flags but I still have problems that I am not sure how to resolve.

The link is https://github.com/kclyu/rpi-webrtc-streamer/issues/98#issuecomment-1093111071 and also worth mentioning is I have managed to build v0.74 so far. Only the latter commits, I have issues and tried many things and not sure what to try.