kclyu/rpi-webrtc-streamer #136: Cross Compilation Error

theswedishdeveloper, 2022-03-06:
Hey,

I am trying to build the rpi-webrtc-streamer from source but I have encountered issue while trying to build webrtc from souce. I have followed the steps at https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md but when I am supposed to execute command "gn gen out/arm_build" I can get following error code:

```

ERROR at //build/config/linux/pkg_config.gni:104:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /root/Workspace/webrtc/src/out/arm_build/
Command: python3 /root/Workspace/webrtc/src/build/config/linux/pkg-config.py -s /opt/rpi_rootfs/rootfs -a arm --system_libdir lib glib-2.0 gmodule-2.0 gobject-2.0 gthread-2.0
Returned 1.
stderr:

Package glib-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `glib-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'glib-2.0' found
Package gmodule-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gmodule-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gmodule-2.0' found
Package gobject-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gobject-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gobject-2.0' found
Package gthread-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gthread-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gthread-2.0' found
Traceback (most recent call last):
  File "/root/Workspace/webrtc/src/build/config/linux/pkg-config.py", line 248, in <module>
    sys.exit(main())
  File "/root/Workspace/webrtc/src/build/config/linux/pkg-config.py", line 143, in main
    prefix = GetPkgConfigPrefixToStrip(options, args)
  File "/root/Workspace/webrtc/src/build/config/linux/pkg-config.py", line 81, in GetPkgConfigPrefixToStrip
    prefix = subprocess.check_output([options.pkg_config,
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'glib-2.0', 'gmodule-2.0', 'gobject-2.0', 'gthread-2.0']' returned non-zero exit status 1.

See //build/config/linux/BUILD.gn:59:3: whence it was called.
  pkg_config("glib") {
  ^-------------------
See //build/config/compiler/BUILD.gn:274:18: which caused the file to be included.
    configs += [ "//build/config/linux:compiler" ]
                 ^------------------------------
```

Does anyone know how to solve it?

I am using Ubuntu 20.04.4 LTS and are building for Raspberry pi 3 B+.

Thanks in advance!

@kclyu

theswedishdeveloper, 2022-03-07:
I think the problem is caused because the "instructions" are outdated. Please can someone who managed cross-compile successfully share with me here, what versions of webrtc you use and what version of raspios to use to create rootfs. Thank you.

samyzhh, 2022-03-15:
I have the same problem as you, don't know how to solve it?

----------------------------------------------------------------
ERROR at //build/config/linux/pkg_config.gni:104:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/samy/Workspace/webrtc/src/out/arm_build/
Command: python3 /home/samy/Workspace/webrtc/src/build/config/linux/pkg-config.py -s /opt/rpi_rootfs/rootfs -a arm --system_libdir lib gmodule-2.0 gthread-2.0 gtk+-3.0
Returned 1.
stderr:

Package gmodule-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gmodule-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gmodule-2.0' found
Package gthread-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gthread-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gthread-2.0' found
Package gtk+-3.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gtk+-3.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gtk+-3.0' found
Traceback (most recent call last):
  File "/home/samy/Workspace/webrtc/src/build/config/linux/pkg-config.py", line 248, in <module>
    sys.exit(main())
  File "/home/samy/Workspace/webrtc/src/build/config/linux/pkg-config.py", line 143, in main
    prefix = GetPkgConfigPrefixToStrip(options, args)
  File "/home/samy/Workspace/webrtc/src/build/config/linux/pkg-config.py", line 81, in GetPkgConfigPrefixToStrip
    prefix = subprocess.check_output([options.pkg_config,
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'gmodule-2.0', 'gthread-2.0', 'gtk+-3.0']' returned non-zero exit status 1.

See //build/config/linux/gtk/BUILD.gn:13:1: whence it was called.
pkg_config("gtk_internal_config") {
^----------------------------------
See //examples/BUILD.gn:719:17: which caused the file to be included.
      deps += [ "//build/config/linux/gtk" ]
                ^-------------------------